JarkkoPFC/arduino-music-player #6: Wishlist: Linux gcc compilation support for pmf_converter?

sta-c0000, 2019-12-09:
Wishlist: Linux gcc compilation support for pmf_converter

I see there is mention of Linux platform support in the `sxp_src` tree files.  Tried compiling with `gcc` using the `PFC_RELEASE PFC_COMPILER_GCC PFC_PLATFORM_LINUX64` options without luck.  Appears might be more work required than a few missing includes (e.g. `<sched.h>` & `<unistd.h>`) and what not.

What are your thoughts?

BTW, `pmf_player` is very cool, thank you.  It worked _out of the box_ (as is) on an **Adafruit Circuit Playground Express**.  [Here's a quick video](https://gofile.io/?c=dRFmYA) connected to battery and using tiny on-board speaker.  The "_neopixels_" are using a simple `.setPixelColor` library call in `example_visualization`.

Thanks!

JarkkoPFC, 2019-12-10:
Glad to hear the player works for you!

Thanks for pointing out the Linux issues with the converter. I'll have a look.

JarkkoPFC, 2019-12-10:
Should work now (tested on "gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0")

closed 2019-12-10 by JarkkoPFC

sta-c0000, 2019-12-10:
Brilliant, worked perfectly!
Thank you.