JarkkoPFC/arduino-music-player #11: AMP as a Uzebox sound card?

danboid, 2023-11-07:
Hi Jarkko

Did you already know about the open source 8 bit games console the Uzebox?

https://uzebox.org/

It only has 4K RAM and part of that gets used by the Uzebox sound engine, which doesn't currently have MOD playback/

Do you think it would be possible to trigger a sample or two over the MOD/SID or whatever is being played by AMP, just using one Teensy or similar? Maybe have the MOD playing in the left and the samples in the right channel?

Could it be possible to hook the AMP up to the Uzebox AVR MCU via SPI or UART and have it function as a simple soundcard for playing MOD files and samples for SFX?

Which MCU would you recommend for this purpose, if you think it could work?

Thanks!

JarkkoPFC, 2023-11-13:
You can have AMP playing a MOD and programmatically trigger the samples (with given note, volume & effect) included in the MOD on the channels. For example, if you have a 4 channel MOD, you could allocate an extra channel in a tracker for the MOD and then trigger samples in that extra channel to avoid messing with the music.

danboid, 2023-11-13:
Absolutely you could yes! How did that totally obvious use not occur to me? Ha!

JarkkoPFC, 2023-11-13:
Not that I'm aware of. You can check the [Making Electronic Instruments](https://github.com/JarkkoPFC/arduino-music-player#making-electronic-instruments) section in the readme for some details.

closed 2023-11-24 by JarkkoPFC